hn/bsh-home-appliances #1: Emulating a BSH extractor hood

dosordie, 2024-03-13:
Hello, I don't know if I'm in the right place, but I'll try my luck.
My induction hob can control a BSH extractor hood, but you have to pair the devices.
I'm looking for a way to pretend that a Siemens cooktop has a WiFi extractor hood.
The aim would then be to be able to switch the ventilation levels and the lighting from the cooktop.
About node Red, iobroker etc. I could then control the other devices
Greeds,
Dominik

hn, 2024-03-13:
> My induction hob can control a BSH extractor hood, but you have to pair the devices.

How is the pairing done? Via WiFi? Do you have to connect both of them to your home WiFi or does one of the devices act as a WiFI access point and the other one as a client? Do you need to use the Home Connect cloud for pairing?

dosordie, 2024-03-15:
I think the paring is done via lokal Network. 
I captured the Wifi Packets via Wireshark. 
The cooktop sends mdns requests via Broadcast when it is in paring mode. I can post the capture later.
But i dont have a extractor hood so i dont know how the answer has to be.

hn, 2024-11-01:
Stale topic, please add more information and/or re-open issue if necessary.

closed 2024-11-01 by hn