hn/bsh-home-appliances #3: Test with Model Bosch WAQ24468/09ES

akdri, 2024-09-16:
Hello, first, thanks for your work.

This weekend I decided to connect my washing machine with an esp01 and use your code.

My washing machine is another model WAQ24468/09ES so I had to make several changes to the code for my model as it differs in functions and features as well as set up all the programs on my washing machine.

I also think I've discovered the function of:
14.10-08 00 ?
In my washing machine
14.10-08 00 Spin enabled
14.10-08 01 Spin disabled

This parameter is always sent when the spin rpm is changed, I currently have the problem that when I disable the spin of my washing machine for 0 rpm it sends the same code as for 1200 rpm with the difference of change of 14.10-08 01, for this reason for the calculation of rpm they should take into account the value of this bit and if it is 1 the rpm should be 0 rpm.

For those who know how to program in ESPHOME it should be very simple to create another variable that checks this bit and if it is one the rpm should be 0 because the spin is disabled.
It would be necessary to modify the calculation of rpm with rpm * (1 - Bit of 1008)

I am a newbie in ESPHOME if I can do it I will attach the code if not I suppose it will be simple for you to do it.

I am attaching my code in case you want to use it as an example for my model and if it helps someone else.

# !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!CUSTOM MODEL WAQ24468/09ES
substitutions:
  devicename: lavadora
  upper_devicename: Lavadora Bosch

esphome:
  name: $devicename
  friendly_name: $upper_devicename

external_components:
  - source: github://hn/bsh-home-appliances@master

esp8266:
  board: esp01_1m

logger:

api:
  encryption:
    key: !secret api_key
ota:
- platform: esphome
  password: !secret ota_pass

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 10.5dB
  manual_ip:
    static_ip: !secret wifi_Lavadora
    gateway: !secret wifi_gateway
    subnet: !secret wifi_subnet

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lavadora AP"
    password: !secret wifi_password

captive_portal:

uart:
  id: dbus_uart
  rx_pin: GPIO14
  baud_rate: 9600

bshdbus:
  uart_id: dbus_uart

binary_sensor:
  - platform: bshdbus
    dest: 0x14
    command: 0x1007
    binary_sensors:
      - id: bsh_wm_feat_waterplus
        name: Extra
        icon: mdi:water-plus
        lambda: return (x[0] >> 1) & 0x01;
      - id: bsh_wm_feat_eco
        name: Eco Perfect
        icon: mdi:sprout-outline
        lambda: return (x[0] >> 5) & 0x01;
      - id: bsh_wm_feat_prewash
        name: Rápido
        icon: mdi:clock-fast
        lambda: |-
          int sensor_value = x[0];
            int mask = 0b01100000; // Máscara para los bits en las posiciones 6 y 7
            if ((sensor_value & mask) == 0) {
                // Los bits en las posiciones 6 y 7 están a cero
                return 1; // O cualquier valor que indiques para el caso positivo
            } else {
                // Al menos uno de los bits está en 1
                return 0; // O cualquier valor que indiques para el caso negativo
            }
      - id: bsh_wm_feat_anticrease
        name: Fácil Plancha
        icon: mdi:iron
        lambda: return (x[1] >> 7) & 0x01;
  - platform: bshdbus
    dest: 0x15
    command: 0x1100
    binary_sensors:
      - id: bsh_wm_start_button
        name: Start
        icon: mdi:button-pointer
        lambda: return 1;
        on_press:
          - delay: 200ms
          - lambda: id(bsh_wm_start_button).publish_state(NAN);
  - platform: bshdbus
    dest: 0x26
    command: 0x1701
    binary_sensors:
      - id: bsh_wm_program_started
        name: Lavando
        entity_category: diagnostic
        icon: mdi:ray-start-arrow
        lambda: return 1;
        on_press:
          - delay: 1s
          - lambda: id(bsh_wm_program_started).publish_state(NAN);

sensor:
  - platform: bshdbus
    dest: 0x14
    command: 0x1004
    sensors:
      - id: bsh_wm_temperature
        name: Temperatura
        device_class: temperature
        state_class: measurement
        unit_of_measurement: °C
        accuracy_decimals: 0
        lambda: return x[0];
        filters:
          - calibrate_linear:
             method: exact
             datapoints:
              - 0 -> 0.0
              - 1 -> 20.0
              - 2 -> 30.0
              - 3 -> 40.0
              - 4 -> 60.0
              - 5 -> 70.0
              - 6 -> 80.0
              - 7 -> 90.0
  - platform: bshdbus
    dest: 0x14
    command: 0x1006
    sensors:
      - id: bsh_wm_rpm
        name: Centrifugado
        device_class: speed
        state_class: measurement
        unit_of_measurement: rpm
        accuracy_decimals: 0
        lambda: return x[0];
        filters:
          - multiply: 10
  - platform: bshdbus
    dest: 0x17
    command: 0x4010
    sensors:
      - id: bsh_wm_unbalance_x
        name: Desequilibrio X
        entity_category: diagnostic
        icon: mdi:axis-x-rotate-clockwise
        state_class: measurement
        unit_of_measurement: m/s²
        accuracy_decimals: 0
        lambda: return (int16_t) ((x[1] << 8) | x[2]);
        filters:
          - throttle: 3s
          - timeout: 90s
      - id: bsh_wm_unbalance_z
        name: Desequilibrio Z
        entity_category: diagnostic
        icon: mdi:axis-z-rotate-clockwise
        state_class: measurement
        unit_of_measurement: m/s²
        accuracy_decimals: 0
        lambda: return (int16_t) ((x[3] << 8) | x[4]);
        filters:
          - throttle: 3s
          - timeout: 90s
      - id: bsh_wm_unbalance_y
        name: Desequilibrio Y
        entity_category: diagnostic
        icon: mdi:axis-y-rotate-clockwise
        state_class: measurement
        unit_of_measurement: m/s²
        accuracy_decimals: 0
        lambda: return (int16_t) ((x[5] << 8) | x[6]);
        filters:
          - throttle: 3s
          - timeout: 90s
  - platform: bshdbus
    dest: 0x26
    command: 0x1020
    sensors:
      - id: bsh_wm_washmodule
        name: Ciclo Lavado
        entity_category: diagnostic
        icon: mdi:map-marker-path
        accuracy_decimals: 0
        unit_of_measurement: ciclo
        lambda: return x[0];
  - platform: bshdbus
    dest: 0x2a
    command: 0x1600
    sensors:
      - id: bsh_wm_remain
        name: Restante
        device_class: duration
        state_class: measurement
        unit_of_measurement: min
        accuracy_decimals: 0
        lambda: return x[0];

text_sensor:
  - platform: bshdbus
    dest: 0x14
    command: 0x1005
    text_sensors:
      - id: bsh_wm_program
        name: Programa
        icon: mdi:numeric
        lambda: return std::to_string(x[2]);
        filters:
          - map:
            - 0 -> Apagado
            - 1 -> Algodón
            - 2 -> Algodón +
            - 3 -> Sintéticos
            - 4 -> Sintéticos +
            - 5 -> Mix
            - 6 -> Delicado
            - 7 -> Lana
            - 8 -> Centrifugado
            - 9 -> Desagüe
            - 10 -> Rapido 15
            - 11 -> Ropa Oscura
            - 12 -> Camisas/Blusas
            - 13 -> Plumas
            - 14 -> Sport
            - 15 -> Antialergias
  - platform: bshdbus
    dest: 0x26
    command: 0x1200
    text_sensors:
      - id: bsh_wm_door
        name: Puerta
        icon: mdi:door
        lambda: return std::to_string(x[0]);
        filters:
          - map:
            - 0 -> Cerrada
            - 1 -> Bloqueada
            - 2 -> Abierta

hn, 2024-09-16:
Thanks for contributing.

I suggest to keep the `bshdbus` sensors simple, just do basic bit-shifting/masking there.

If you want to combine multiple values, use a [template sensor](https://esphome.io/components/sensor/template.html):

```
sensor:
  - platform: template
    name: "My magic calculated something"
    lambda: return id(some_sensor).state * id(some_other_sensor).state // your logic here
```

I would be pleased if you submit your yaml (e.g. as `bsh-dbus-waq24468-09es.yaml`) to this project, just create a pull request.

akdri, 2024-09-17:
Good morning,

Thanks for your help.

I have little time to do tests and changes, but when I finish fixing some things in the code, I will send you my yaml file, so you can publish it for my model. I'm sure you can also optimize some of my code, because I'm a newbie at programming in Esphome.

I'm also trying to probe if I can extract parameters such as real temperature of the probe, water level, motor speed, direction of rotation, etc.

In the case of my washing machine I find a lot of traffic on destination addresses 0x16

I think there may be relevant information in

0x16 0X3015

16.30-15 0100 72 XX XX 72 YY YY (Y= 00 00 or 03 20)
16.30-15 0103 6C XX XX 6C 03 20
16.30-15 0104 72 XX XX 72 YY YY
16.30-15 0107 6C XX XX 6C YY YY

The values ​​with XX vary constantly some in a few seconds and others every minute.

I'm trying to make a sensor to capture these values ​​and try to find a relationship with the parameters shown, but I don't know if my code is correct:

  - platform: bshdbus
    dest: 0x16
    command: 0x3015
    sensors:
      - id: bsh_wm_test
        name: test1
        entity_category: diagnostic
        icon: mdi:map-marker-path
        accuracy_decimals: 0
        unit_of_measurement: u
        lambda: |-
            if (x[1] == 0x03) {
                return (int16_t) x[3] | x[4];
            } else {
                return NAN;
            }

Also I created another sensor to know the status of the washing machine, on its page this operation is described, but in the example the sensor does not exist:

  - platform: bshdbus
    dest: 0x26
    command: 0x1101
    text_sensors:
      - id: bsh_wm_state
        name: Estado
        icon: mdi:state-machine
        lambda: return std::to_string(x[0]);
        filters:
          - map:
            - 0 -> Finalizada
            - 1 -> Acabando
            - 2 -> Lavando

hn, 2024-09-18:
> I think there may be relevant information in
>
> 0x16 0X3015
>
> 16.30-15 0100 72 XX XX 72 YY YY (Y= 00 00 or 03 20)
> 16.30-15 0103 6C XX XX 6C 03 20
> 16.30-15 0104 72 XX XX 72 YY YY
> 16.30-15 0107 6C XX XX [...]

If I remember correctly I've never observed any 16.30-15 frames so I can not add any useful information. Y=0x320 (= decimal 800) looks like a human-set value, e.g. for rpm or timeout.

> Also I created another sensor to know the status of the washing machine, on its page this operation is described, but in the example the sensor does not exist:
> 
> platform: bshdbus
> dest: 0x26
> command: 0x1101

I didn't add this sensor because I wasn't sure if the information was correct. Happy if it works for you after thorough analysis.

akdri, 2024-09-23:
Good morning.
After many tests I can confirm that:
0x16 3015 is the inverter's response

I am attaching everything I have compiled in case you want to add it to the documentation. I hope it helps you.

A typical communication is

dest 0x31 cmd 0x3001: 0x7200000000 //Request to the drive to stop and turn right CC DD EE FF GG HH
dest 0x16 cmd 0x3002: 0x0107 //Request to the drive? AA BB
dest 0x31 cmd 0x3014: 0x
dest 0x16 cmd 0x3015: 0x01076c02f46c02ec // Immediate response from the drive with the current status (in this case it says I am turning left at speed 02f4 of the programmed speed 02ec
dest 0x31 cmd 0x3014: 0x
dest 0x16 cmd 0x3015: 0x0100720000720000 // the speed controller responds again after the command is executed (I have changed the rotation to the left and I am at speed 0)

dest 0x16 cmd 0x3015: 0xAABBCCDDEEFFGGHH

AA is always 01
BB I am not clear about its behavior, they seem to be orders related to the time that the requested speed will last and stop orders, but they vary according to the wash cycle and I do not quite understand the logic of this Byte
CC is the current rotation direction of the drum (although this stopped should indicate the direction of rotation, in the speed controllers it should always be indicated) 6C right 72 left
DD and EE are the current rpm are calculated (DDEE/16) (I am 100% sure)

FF The programmed rotation direction (it should always be equal to CC if the speed controller does not have problems)

GGHH are the target rpm are calculated (GGHH/16) (I am 100% sure)

I have tried to see if there were other parameters exposed by the bus such as the status of the water valves, the drain, the status of the hot water resistance, the temperature of the temperature probe and the water level of the pressure switch sensor, but it seems that these data are not exposed to the bus.

Since they are controlled directly by the central unit, there is no reason for them to be exposed to bus, because they are processed internally.

I suppose that there are commands to make a query of the status of these parameters, surely BSH must have diagnostic tools to be able to verify these parameters.

But without someone who knows these commands it will be impossible to access these parameters.

The code for the rpm sensors are:

  - platform: bshdbus
    dest: 0x16
    command: 0x3015
    sensors:

    // Para el byte x[3] y x[4] son rpm/16
      - id: bsh_wm_rpm_real
        name: rpm actuales
        accuracy_decimals: 0
        device_class: speed
        unit_of_measurement: rpm
        lambda: return (((int16_t) x[3] << 8 | x[4])/16);

    // Para el byte x[6] y x[7] 
      - id: bsh_wm_rpm_target
        name: rpm objectivo
        accuracy_decimals: 0
        device_class: speed
        unit_of_measurement: rpm
        lambda: return (((int16_t) x[6] << 8 | x[7])/16);
        
// Sensor right and left:

      - id: bsh_wm_turn_right
        name: Turn right
        accuracy_decimals: 0
        unit_of_measurement: r
        lambda: return (x[5] >> 3) & 0x01;

      - id: bsh_wm_turn_left
        name: Turn left
        accuracy_decimals: 0
        unit_of_measurement: r
        lambda: return (x[5] >> 4) & 0x01;
        
        
[image]

hn, 2024-09-23:
Hey, that's great! Have to check that in detail later.

So the D=3 of the "DS" byte is the "motor controller", that part was always missing on my side (in my [setup](https://github.com/hn/bsh-home-appliances/blob/master/bsh-wm-lab-setup.png) I do not have a motor anymore).

Since 0x6C is ASCII "l" and 0x72 is "r" you probably could do something like this:

```
    text_sensors:
      - id: bsh_wm_rotdirection
        name: Direction of rotation
        entity_category: diagnostic
        icon: mdi:crop-rotate
        lambda: return std::to_string(x[5]);
        filters:
          - map:
            - r -> Right
            - l -> Left
```

akdri, 2024-09-23:
[logs_lavadora.txt](https://github.com/user-attachments/files/17094642/logs_lavadora.txt)
Yes, once the tests are finished I will convert the sensor to text as you have indicated, but to do the tests it is always more convenient to use 1 and 0 to be able to show the data in a graph.

I am sending you my yaml which is still being tested

and a log of my washing machine is not complete, but this way you will be able to observe the traffic of the inverter.

akdri, 2024-09-23:
Where can I attach the yaml? Through the thread it doesn't let me

ok, I send  (zip)
[bsh-dbus-WAQ24468.zip](https://github.com/user-attachments/files/17094731/bsh-dbus-WAQ24468.zip)

hn, 2024-09-23:
@akdri I pushed your draft as https://github.com/hn/bsh-home-appliances/commit/fdf7a1fa85dc1070d264e6442d278ba4b1df98df, *please* use Github's edit and pull request functions to submit changes to this file.

hn, 2024-11-01:
Please re-open and/or submit PR if there is any new information

closed 2024-11-01 by hn