hn/bsh-home-appliances #6: Getting useful information from Bosch WAE284F0NL/05

Thijn, 2024-10-25:
Hi,

Awesome work on reverse engineering so far!

I'm having trouble getting my washing machine to output useful information.
I hooked up an ESP with the logging sketch, and seem to get quite some information but none of them seem to match the control commands listed in this repo.

Below is the log when I start up the machine, and keep rotating the program select knob.
The 23.10-02 commands seem to be sent when I turn the knob, but they aren't unique so don't really mean much.
The rest is just 22.10-03 commands, which does seem to contain some information about what's happening, but it doesn't make much sense as to what it means.

Have you seen anything like this before? Or do you have any pointers as to how to debug this problem?

Any help is much appreciated.

`dB/S/H/ D-Bus logger started.
54 | (read=timout)
02 | 11.10-05 | ca ed (crc=ok) | (ack=err, re-evaluate byte)
02 | 11.10-05 | ca ed (crc=ok) | (ack=err, re-evaluate byte)
02 | 11.10-05 | ca ed (crc=ok) | (ack=err, re-evaluate byte)
02 | 11.10-05 | ca ed (crc=ok) | (ack=err, re-evaluate byte)
02 | 11.10-05 | ca ed (crc=ok) | (ack=none)
03 | 24.10-06 | 02          | da ea (crc=ok) | 2a (ack=ok)
03 | 14.10-07 | 82          | 54 ba (crc=ok) | 1a (ack=ok)
02 | 11.10-05 | ca ed (crc=ok) | 1a (ack=ok)
0a | 22.10-03 | 00 00 00 00 00 00 00 02 | 8e 1b (crc=ok) | 2a (ack=ok)
0c | 21.10-00 | 00 00 00 00 00 00 00 02 fe 01 | 07 1a (crc=ok) | 2a (ack=ok)
04 | 23.10-02 | 00 04       | d5 84 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 40 40 00 00 00 00 00 02 | a5 27 (crc=ok) | 2a (ack=ok)
04 | 23.10-02 | 00 04       | d5 84 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 04 40 00 00 00 00 00 02 | 5c 9a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 04 40 00 00 00 00 00 02 | 5c 9a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 00 00 00 00 00 02 | 7e c4 (crc=ok) | 2a (ack=ok)
04 | 23.10-02 | 00 04       | d5 84 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 00 00 00 00 10 02 | 7d b7 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 00 02 | 68 df (crc=ok) | 2a (ack=ok)
04 | 23.10-02 | 00 04       | d5 84 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 7c 77 10 02 | 6b ac (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 10 02 | e3 29 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 00 02 | e0 5a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 10 02 | e3 29 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 00 02 | e0 5a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 10 02 | e3 29 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 00 02 | e0 5a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 10 02 | e3 29 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 00 02 | e0 5a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 10 02 | e3 29 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 05 01 60 60 69 5f 00 02 | e0 5a (crc=ok) | 2a (ack=ok)
[... 143 lines cut ...]
0a | 22.10-03 | 20 08 60 77 60 5d 10 02 | 05 aa (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 60 5d 00 02 | 06 d9 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 60 5d 00 02 | 06 d9 (crc=ok) | 2a (ack=ok)
04 | 23.10-02 | 00 04       | d5 84 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 60 5d 10 02 | 05 aa (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 7c 7d 10 02 | d7 f9 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 7c 7d 00 02 | d4 8a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 7c 7d 10 02 | d7 f9 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 7c 7d 00 02 | d4 8a (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 20 08 60 77 7c 7d 10 02 | d7 f9 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 08 60 77 7c 7d 10 02 | 91 25 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 08 60 77 7c 7d 10 02 | 91 25 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 77 7c 7d 10 02 | a3 ad (crc=ok) | 2a (ack=ok)
04 | 23.10-02 | 00 04       | d5 84 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 77 7c 7d 00 02 | a0 de (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 60 3e 5d 00 02 | c6 62 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 60 3e 5d 10 02 | c5 11 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 60 3e 5d 00 02 | c6 62 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 60 3e 5d 10 02 | c5 11 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 60 3e 5d 00 02 | c6 62 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 10 06 60 60 3e 5d 10 02 | c5 11 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 00 00 00 00 00 00 00 02 | 8e 1b (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 00 00 00 00 00 00 10 02 | 8d 68 (crc=ok) | 2a (ack=ok)
0a | 22.10-03 | 00 00 00 00 00 00 00 02 | 8e 1b (crc=ok) | 2a (ack=ok)
ff | `

hn, 2024-10-26:
Since I only have one (n=1) washing machine accessible, all my findings are based on this single machine.

A first „wild“ guess is that your machine _might_ have more than one D-Bus and you‘re only listening to a limited part of the communication. 23.10-02 is a command sent **to** the front panel (D=2) and therefore this might be just the ACK for the real knob-command which has been sent elsewhere.

Thijn, 2024-10-26:
Thanks for the quick reply.

That does make sense. I added my ESP between what I thought was the control board and display board.
Upon looking at spare parts, what I thought was a control board is actually called Power module. The display is called the control board itself. The display only has one connector thought, which goes to the power module . And that module has multiple connectors connecting to the motor and such.

Where is your ESP plugged in? Is there an unused connector you used, or did you splice an existing connection (Like I did)

hn, 2024-10-26:
Connector 9, 10 or 11 of the [control board](https://github.com/hn/bsh-home-appliances?tab=readme-ov-file#control-board-epw66018). On my board these connectors are physically connected.

Having more than one D-Bus would be sonewhat surprising since it counteracts the idea of a bus system. But you never know.

Thijn, 2024-10-26:
I've measured all the pins on my control board, and some measure around 5 volts but don't seem to be carrying a D-Bus signal.

The only one carrying a signal is the one coming from to control panel, which is what I have been using. It's a 3 pin connector with 9v VCC, 5v data and GND.

I'll have another look later to see if I can find another source of the D-Bus signal. Thanks a lot for your help so far.

Casandro, 2024-10-26:
AFAIK the D-Bus uses open collector logic, so those 5v probably come from some pull-up resistor. I don't know if you have access to the "Home Connect" Module. it should request a full dump of all the available parameters. Sending commands to the device is done by setting parameters.

Thijn, 2024-10-27:
I don't have access to Home Connect. I don't think my washer even supports that, it's quite old.

Here are some pics of the boards, maybe something stands out for you guys.
I'm more of a software guy, hardware isn't my strong suit.

Control board front (Upside down)
[image]
back
[image]

Control panel front (With the display LED diffuser removed)
[image]
back
[image]

The connector on the control panel is the one I got the dump from in my first post.
Which is connected on the connector on the bottom right (The top one of the 2) of the control board.

hn, 2024-10-28:
- Can you please decode the manufacturing date (FD-Nr) of your machine, see [here](https://github.com/hn/bsh-home-appliances/tree/master#serial-numbers)
- Just out of interest, can you please post the model number of the CPU (square shaped chip on your first image)
- The two connectors on the bottom right of your second image probably both are D-Bus (hard to see on the first image: the traces _seem_ to be interconnected)
- Every time you rotate the knob (`23.10-02` in your log) bytes 3, 4 and 5 of the `22.10-03` frames seem to change, e.g.
  ```
              0  1  2  3  4  5  6  7
  22.10-03 | 05 01 60 60 7c 77 00 02     // likely is 1:30 with LED off
  22.10-03 | 05 01 60 60 7c 77 10 02     // likely is 1:30 with LED on
  22.10-03 | 05 01 60 60 69 5f 00 02     // likely is 1:46
  22.10-03 | 05 01 60 60 7c 5d 00 02     // likely is 1:35
  22.10-03 | 10 08 60 77 5d 5d 00 02     // likely is 0:55
  ```
  Can you please check if these bytes correspond to the remaining time digits X:XX on the user front panel as commented above? Please write down display and frame content otherwise. Bytes 0 and 1 probably relate to other symbols (temperature/speed) on the user front panel.
- Do you observe any D-Bus frames if you press buttons (Anti-crease, Water plus, ...) on the user panel?

As said before "Having more than one D-Bus would be somewhat surprising since it counteracts the idea of a bus system". I do not think that there is more than one D-Bus. Your appliance seems to be of an older machine generation.

Thijn, 2024-10-28:
- I'll have a look at the manufacturing date tomorrow
- CPU has the following text: 
SC100203
CFBE  4L47P
CTEE0631

- Yes, the second connector also carries D-Bus.
- I did come to the same conclusion regarding the 22.10-03 frames.

I do wonder how you determine the remaining time. They are correct, but how do you get 1:46 from those frames?

The first bytes are the temperature, second the RPM. Frame 6 is the Start LED, so it was blinking on most of the log.

Thanks again for putting this repository online in the first place, and the massive help you've given decoding my ancient little washing machine!

hn, 2024-10-29:
- Manfacturing date seems to be 2006 (=> 86+20). This supports the assumption that BSH introduced the D-Bus gradually and devices before 2010 only realized “simple” display tasks via the D-Bus.
- SC100203 seems to be one of the countless Freescale/NXP MCUs.

> I do wonder how you determine the remaining time. They are correct, but how do you get 1:46 from those frames?

Well, reverse engineering is a riddle sometimes. Bytes 3, 4 and 5 look like "random" numbers, right? But they are not. Each bit in these bytes corresponds to a single bar in a 7-digit LED display. If you make an “educated guess” that the first digit of the remaining time is probably a “0” or a “1” and the second digit is from the 0-5 range, you can deduce the numbers with increased probability.

To be honest, I examined a similar device some time ago and so this time I only had to solve a small riddle :)

I pushed https://github.com/hn/bsh-home-appliances/commit/80e7f2549e747e5ebeef4da4c081277519033e56. If you like, test the YAML and do a PR for the missing parts (speed/temperature/...).

commit e2d2c721 (2024-10-31): Merge pull request #8 from Thijn/master, closes #6

WAE284F0NL: Add sensors, test and fix existing sensors
closed 2024-10-31 by hn